urbste/OpenImuCameraCalibrator #1: IPFS Dataset downloading

SKioresku, 2021-04-05:
I'm trying to download the dataset from the link, but it doesn't seem to work.

SKioresku, 2021-04-05:
It was necessary to enter the following command ipfs get QmfRQmJcFu7TN4WFXwAbNMosvwo4heGKQtKbBSLpLPqFF7

closed 2021-04-05 by SKioresku

urbste, 2021-04-05:
I actually also changed the link to a google drive download link.

urbste, 2021-04-05:
Let me know if the calibration works for you or if you face any issues.

SKioresku, 2021-04-05:
> Let me know if the calibration works for you or if you face any issues.
I kept IMU and Images time in microseconds, but i have stange time_offset_gyro_to_cam after EstimateImuToCameraRotation
[imu_rotation_init.txt](https://github.com/urbste/OpenImuCameraCalibrator/files/6258922/imu_rotation_init.txt)

SKioresku, 2021-04-05:
I am using this library with small modifications to calibrate my smartphone.

SKioresku, 2021-04-05:
I changed some multipliers in these functions 
[image]
[image]
[image]

SKioresku, 2021-04-05:
> I actually also changed the link to a google drive download link.

Thanks)