urbste/OpenImuCameraCalibrator #2: Error in continuous_time_imu_to_camera_calibration.cc

SKioresku, 2021-04-15:
[image]

urbste, 2021-04-15:
Looks like an eigen alignment issue. I never had that Problem. What versions are you using? Could you run à RelWithDebug build to find out where it is crashing?

SKioresku, 2021-04-20:
Which compiler did you use? What are the versions of the Eigen and Sophus libraries? I have Sophus 1.1.0 and Eigen  3.3.9. Compiled with GNU 9.3

urbste, 2021-04-21:
Eigen is version 3.3.7 in my case. Sophus is in the third_party library. So there is no need to install it in the system. My gcc version is 8.4.0.

I should probably dockerize the calibration pipeline in the future.

urbste, 2021-04-21:
I just updated my Eigen to 3.3.9. No alignment issues.

SKioresku, 2021-04-21:
What Ubuntu version are you using? Can it be VirtualBox problem?

urbste, 2021-04-21:
I am on Ubuntu 18.04.5. I never tried it in a VirtualBox. 

What you could try is to turn off Eigen alignment by passing EIGEN_MAX_ALIGN_BYTES=0 to the compiler (https://eigen.tuxfamily.org/dox/TopicPreprocessorDirectives.html).
Have a look at line 12 in CMakeLists.txt.

SKioresku, 2021-04-22:
Camera calibration not working with "Double_Sphere" camera model  and new Theia library
[image]

urbste, 2021-04-22:
Can you maybe send me one of your datasets so I can check? I just tries it and it works with the Double Sphere camera model.

SKioresku, 2021-04-22:
Dataset downloaded from your link. Thats my code with hardcoded pathes and corners of camera calibration v
ideo in binary format 
[calibrate_camera.zip](https://github.com/urbste/OpenImuCameraCalibrator/files/6359232/calibrate_camera.zip)

SKioresku, 2021-04-22:
This is source libraries for OpenImuCameraCalibrator building. https://filetransfer.io/data-package/QzB9NRwV#link

urbste, 2021-05-01:
Sorry for my late reply. I ran the calibrate_camera.cc with the corners.uson that you supplied. I get a valid camera calibration without any problems or crashes: 
Final camera calibration reprojection error: 0.295454 from 40 view.
Focal Length:434.714px Principal Point: 479.042/273.613px.
DOUBLE_SPHERE model: XI: -0.00281319 ALPHA: 0.535422

The only thing you can try is doing a debug build and trying to find out where it is crashing.

urbste, 2021-05-01:
The library versions I am using: 
Newest TheiaSfM master from my fork
Eigen 3.3.9
Ceres-solver 2.0.0
OpenCV 4.5.1

SKioresku, 2021-05-13:
Ok, thank you very much

closed 2021-05-13 by SKioresku

laldev, 2021-06-04:
As you mentioned "Sophus is in the third_party library folder. So there is no need to install it in the system."  https://github.com/urbste/OpenImuCameraCalibrator/issues/2#issuecomment-824088319 , but I cant find any files inside Sophus after the git cloning OpenImuCameraCalibrator . What is the corect way to install Sophus?

I have tried to install it manually, but make step of OpenImuCameraCalibrator fails with the following error:

imu_camera_calibrator.cc:(.text._ZN3fmt2v55printIZZN6Sophus7SO3BaseINS2_3SO3IN5ceres3JetIdLi4EEELi0EEEE9normalizeEvENKUlvE_clEvE3strJN5Eigen9TransposeINSC_6MatrixIS7_Li4ELi1ELi0ELi4ELi1EEEEEEEENSt9enable_ifIXsrNS0_8internal9is_stringIT_EE5valueEvE4typeERKSK_DpRKT0_[_ZN3fmt2v55printIZZN6Sophus7SO3BaseINS2_3SO3IN5ceres3JetIdLi4EEELi0EEEE9normalizeEvENKUlvE_clEvE3strJN5Eigen9TransposeINSC_6MatrixIS7_Li4ELi1ELi0ELi4ELi1EEEEEEEENSt9enable_ifIXsrNS0_8internal9is_stringIT_EE5valueEvE4typeERKSK_DpRKT0_]+0x74): undefined reference to `fmt::v5::vprint(fmt::v5::basic_string_view<char>, fmt::v5::format_args)'
collect2: error: ld returned 1 exit status
applications/CMakeFiles/continuous_time_imu_to_camera_calibration.dir/build.make:177: recipe for target 'applications/continuous_time_imu_to_camera_calibration' failed
make[2]: *** [applications/continuous_time_imu_to_camera_calibration] Error 1
CMakeFiles/Makefile2:243: recipe for target 'applications/CMakeFiles/continuous_time_imu_to_camera_calibration.dir/all' failed
make[1]: *** [applications/CMakeFiles/continuous_time_imu_to_camera_calibration.dir/all] Error 2
Makefile:90: recipe for target 'all' failed

urbste, 2021-06-05:
You are absolutely right. Thanks for pointing it out. I fixed it and put Sophus in the third_party folder. 
Could you please try and out and see if it works now?

laldev, 2021-06-07:
Hi,  Thanks for your response.
I am getting build error due to the new commit.
ceres_calib_split_residuals.h:497:68: error: ‘const class theia::Track’ has no member named ‘RefBearing’
     const Eigen::Vector3d bearing_d = image_data->Track(track_id)->RefBearing();

I did not dig into the repo as I was just interested to quickly test it if things are already working. It would be great if you could provide a tag with no build errors. Thank you for the great work. I really appreciate the hard work you put in.

urbste, 2021-06-07:
Should be fixed now. I renamed a function in TheiaSfM. I also added a commit tag for TheiaSfM.

laldev, 2021-06-08:
Hi, 
It still gives the same error as before.
I found const Eigen::Vector3d &ReferenceBearingVector() const; in line 82 or track.h
is this what you mena to call here --> const Eigen::Vector3d bearing_d = image_data->Track(track_id)->RefBearing();

urbste, 2021-06-08:
You also have pull and rebuild and install Theia library.

urbste, 2021-06-08:
Yes extactly. But I updated the file yesterday. So make sure to pull the newest changes:
https://github.com/urbste/OpenImuCameraCalibrator/blob/master/include/OpenCameraCalibrator/basalt_spline/ceres_calib_split_residuals.h#L498