urbste/OpenImuCameraCalibrator #3: SyntaxError: Unexpected token = while running extract_metadata.js

laldev, 2021-06-08:
Hi,
Since I have no knowledge in JS, I am finding it hard to debug this error:
```
node $js_path/extract_metadata.js $base_path GH010164.MP4 $base_path 
/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/javascript/helpers/mp4reader.js:96
    onSamples = (id, user, samples) => {
              ^

SyntaxError: Unexpected token =
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:616:28)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/javascript/extract_metadata.js:5:25)
```

I have already installed nodejs(v8.10.0) though apt and did npm install in the directory OpenImuCameraCalibrator/javascript/

This same error is stopping python/run_gopro_calibration.py from executing.

```
==================================================================
Extracting GoPro telemetry for imu bias and camera imu calibration.
==================================================================
/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/javascript/helpers/mp4reader.js:96
    onSamples = (id, user, samples) => {
              ^

SyntaxError: Unexpected token =
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:616:28)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/javascript/extract_metadata.js:5:25)
/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/javascript/helpers/mp4reader.js:96
    onSamples = (id, user, samples) => {
              ^

SyntaxError: Unexpected token =
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:616:28)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/javascript/extract_metadata.js:5:25)
==================================================================
Telemetry extraction took 0.28s.
==================================================================
Traceback (most recent call last):
  File "python/run_gopro_calibration.py", line 315, in <module>
    main()
  File "python/run_gopro_calibration.py", line 200, in main
    telemetry_conv.convert_gopro_telemetry_file(gopro_telemetry, gopro_telemetry_gen)
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/telemetry_converter.py", line 217, in convert_gopro_telemetry_file
    input_telemetry_json, skip_seconds=skip_seconds)
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/telemetry_converter.py", line 58, in read_gopro_telemetry
    self.telemetry = self._read_gopro_telemetry(path_to_jsons, skip_seconds=skip_seconds)
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/telemetry_converter.py", line 68, in _read_gopro_telemetry
    json_file = open(path_to_json, 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/home/EU/USER/work/vendor/data/gopro/GoPro9/dataset3/cam_imu/GH010164.json'

```

System: Kubuntu 18.04, 64-bit

urbste, 2021-06-09:
It seems you have installed a fairly old version.
I do have node 12.2.1 LTS with npm 6.14.12 installed.

Could you please try to update and see if it works:
```bash
curl -sL https://deb.nodesource.com/setup_12.x | sudo -E bash -
sudo apt-get install -y nodejs
```

If that fixes the issue I will update the Readme accordingly.

laldev, 2021-06-10:
Yes, it solved the issue.
Installing through the curl command is a mess if you are behind a proxy. The easiest way is to install through binaries.
I did the following:
Downloaded nodejs v12.x binary file from https://nodejs.org/dist/latest-v12.x/  and 
```
cd /usr/local  
tar --strip-components 1 -xzf /path/node-v12.22.1-linux-x64.tar.gz 
```

Few errors that I encountered:
1. Encoding error in python/sew.py line 18.  similar to this [StackOverflow issue](https://stackoverflow.com/questions/18078851/syntaxerror-of-non-ascii-character )
Solution: added the line to the top of the file:  `# -*- coding: utf-8 -*- `

2. Python dependencies that had to be installed
`pip3 install natsort`

3. Suggestion to edit readme: 
Command for running python/run_gopro_calibration.py requires additional parameters `--path_to_build` and `--path_to_src`  has to be specified otherwise they point to another default location.

4. Finally this
```
==================================================================
Estimating IMU biases.
==================================================================
Traceback (most recent call last):
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/get_imu_biases.py", line 5, in <module>
    from sew import knot_spacing_and_variance
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/sew.py", line 145
    def find_uniform_knot_spacing_spectrum(Xhat, times, quality, *, verbose=False, min_dt=None, max_dt=None):
                                                                  ^
SyntaxError: invalid syntax
==================================================================
IMU bias estimation took 0.05s.
==================================================================
==================================================================
Estimating camera poses for IMU - CAM calibration.
==================================================================
Could not open: /home/EU/USER/work/vendor/data/gopro/GoPro9/dataset3/cam/cam_calib_GH010162_ph_2.0.json
F0610 10:50:06.824108 20661 estimate_camera_poses_from_checkerboard.cc:52] Check failed: read_camera_calibration(FLAGS_camera_calibration_json, camera, fps) Could not read camera calibration: /home/EU/USER/work/vendor/data/gopro/GoPro9/dataset3/cam/cam_calib_GH010162_ph_2.0.json
*** Check failure stack trace: ***
    @     0x7f549a1260cd  google::LogMessage::Fail()
    @     0x7f549a127f33  google::LogMessage::SendToLog()
    @     0x7f549a125c28  google::LogMessage::Flush()
    @     0x7f549a128999  google::LogMessageFatal::~LogMessageFatal()
    @     0x556e9f6bfdab  (unknown)
    @     0x7f5497f3ebf7  __libc_start_main
    @     0x556e9f6bf6aa  (unknown)
==================================================================
Pose estimation estimation took 0.98s.
==================================================================
==================================================================
Estimating Spline error weighting and knot spacing.
==================================================================
Traceback (most recent call last):
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/get_sew_for_dataset.py", line 5, in <module>
    from sew import knot_spacing_and_variance
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/sew.py", line 145
    def find_uniform_knot_spacing_spectrum(Xhat, times, quality, *, verbose=False, min_dt=None, max_dt=None):
                                                                  ^
SyntaxError: invalid syntax
==================================================================
Spline weighting and knot spacing estimation took 0.04s.
==================================================================
==================================================================
Initializing IMU to camera rotation.
==================================================================
E0610 10:50:06.887715 20663 reconstruction_reader.cc:65] Could not open the file: /home/EU/USER/work/vendor/data/gopro/GoPro9/dataset3/cam_imu/pose_calib_GH010164.calibdata for reading.
F0610 10:50:06.887758 20663 estimate_imu_to_camera_rotation.cc:55] Check failed: theia::ReadReconstruction(FLAGS_input_pose_calibration_dataset, &pose_dataset) 
*** Check failure stack trace: ***
    @     0x7f734f0fc0cd  google::LogMessage::Fail()
[... 6 lines cut ...]
==================================================================
Spline weighting and knot spacing estimation took 0.01s.
==================================================================
==================================================================
Optimizing IMU to Camera calibration using Spline Fusion.
==================================================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0610 10:50:06.922936 20664 reconstruction_reader.cc:65] Could not open the file: /home/EU/USER/work/vendor/data/gopro/GoPro9/dataset3/cam_imu/pose_calib_GH010164.calibdata for reading.
F0610 10:50:06.922981 20664 continuous_time_imu_to_camera_calibration.cc:87] Check failed: theia::ReadReconstruction(FLAGS_input_pose_dataset, &pose_dataset) Could not read Reconstruction file.
*** Check failure stack trace: ***
==================================================================
Spline weighting and knot spacing estimation took 0.04s.
==================================================================
==================================================================
[... cut ...]
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/print_result_stats.py", line 75, in <module>
    main()
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/print_result_stats.py", line 24, in main
    data = read_calib_json(args.path_results)
  File "/home/EU/USER/work/vendor/linux/OpenImuCameraCalibrator/python/print_result_stats.py", line 9, in read_calib_json
    with open(file, 'r') as f:
IOError: [Errno 2] No such file or directory: '/home/EU/USER/work/vendor/data/gopro/GoPro9/dataset3/cam_imu/cam_imu_calib_result_GH010164.json'
```

laldev, 2021-06-10:
Okay, the error 4 was because my default python was Python2.7 . I changed it to 3.6 as follows:

Set python alternatives (setting python3 with higher priority of 20) 

```
sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.6 20  
sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 10  

# To change options manually
sudo update-alternatives --config python
```

closed 2021-06-10 by laldev

CanCanZeng, 2022-04-01:
Hi, I meet error while extract metadata from gopro max .360 file. Have you ever meet this problem before?
The .360 file is .mp4 file also, and has the same metadata as ordinary .mp4 file created by gopro 9.
```

input video path: ~/360/GS070482.360
Output file will be: ~/360/GS070482.360_telemetry.json
Opening ~/360/GS070482.360

<--- Last few GCs --->

[8211:0x3cfc060]    20138 ms: Mark-sweep 2047.5 (2050.7) -> 2047.5 (2051.4) MB, 740.2 / 0.0 ms  (+ 77.5 ms in 21 steps since start of marking, biggest step 4.4 ms, walltime since start of marking 829 ms) (average mu = 0.157, current mu = 0.018) allocation[8211:0x3cfc060]    21493 ms: Mark-sweep 2048.5 (2051.4) -> 2047.5 (2050.9) MB, 1354.7 / 0.0 ms  (average mu = 0.064, current mu = 0.001) allocation failure scavenge might not succeed

<--- JS stacktrace --->

==== JS stack trace =========================================

    0: ExitFrame [pc: 0x140dff9]
Security context: 0x01d27b0808d1 <JSObject>
    1: /* anonymous */ [0x2ab89b8b0211] [/media/zcc/Data3/Project/Study/Tool/OpenImuCameraCalibrator/javascript/node_modules/mp4box/dist/mp4box.all.js:~7355] [pc=0x3ea4b05a59d7](this=0x2499d819a169 <ISOFile map = 0x35555a9ffd69>,0x2ab89b887a59 <JSObject>)
    2: /* anonymous */ [0x2ab89b8b01d1] [/media/zcc/Data3/Project/Study/Tool/OpenImuCameraCalibrator/javascri...

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
 1: 0xa1a640 node::Abort() [node]
 2: 0xa1aa4c node::OnFatalError(char const*, char const*) [node]
 3: 0xb9a9fe v8::Utils::ReportOOMFailure(v8::internal::Isolate*, char const*, bool) [node]
 4: 0xb9ad79 v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, bool) [node]
 5: 0xd57ff5  [node]
 6: 0xd58686 v8::internal::Heap::RecomputeLimits(v8::internal::GarbageCollector) [node]
 7: 0xd64f45 v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::GCCallbackFlags) [node]
 8: 0xd65df5 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [node]
 9: 0xd688ac v8::internal::Heap::AllocateRawWithRetryOrFail(int, v8::internal::AllocationType, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [node]
10: 0xd2f2cb v8::internal::Factory::NewFillerObject(int, bool, v8::internal::AllocationType, v8::internal::AllocationOrigin) [node]
11: 0x107189e v8::internal::Runtime_AllocateInYoungGeneration(int, unsigned long*, v8::internal::Isolate*) [node]
12: 0x140dff9  [node]
```

CanCanZeng, 2022-04-02:
Hi, I solved this problem by enlarge the heap size. https://stackoverflow.com/questions/59965966/ineffective-mark-compacts-near-heap-limit-allocation-failed-javascript-heap-ou