urbste/OpenImuCameraCalibrator #6: EstimateRadialDistUncalibratedAbsolutePose error

gongping108, 2021-06-23:
EstimateRadialDistUncalibratedAbsolutePose  double free or corruption (out)

urbste, 2021-06-23:
Hi, could you maybe provide some more information? 
Are you using the example datasets or your own? Which environment?

gongping108, 2021-06-24:
I use the example datasets. --path_calib_dataset=/media/root/linux/laneline/kitti_cal/GoPro9/dataset3 --checker_size_m=0.021 --image_downsample_factor=2 –camera_model=PINHOLE
environment:
Ubuntu18.04 melodic. 
I cannot find the function EstimateRadialDistUncalibratedAbsolutePose

urbste, 2021-06-24:
Hi. I guess the problem is that you are using --camera_model=PINHOLE.
PINHOLE alone will only model focal length and principal point but no other distortion coefficients.
As you can see in the example dataset, the lines of the checkerboard are not straight anymore and hence you have to use a camera model that models these effects.

So please use any of the other camera models: e.g. try --camera_model=DIVISION_UNDISTORTION

gongping108, 2021-06-24:
I use --camera_model=PINHOLE.

urbste, 2021-06-24:
Please use --camera_model=DIVISION_UNDISTORTION

gongping108, 2021-06-24:
I use the example datasets. --path_calib_dataset=/media/root/linux/laneline/kitti_cal/GoPro9/dataset3 --checker_size_m=0.021 --image_downsample_factor=2 –camera_model=DIVISION_UNDISTORTION
environment:
Ubuntu18.04 melodic.
I cannot find the function EstimateRadialDistUncalibratedAbsolutePose

gongping108, 2021-06-24:
I'm really really really need your help! Thanks

urbste, 2021-06-24:
Did you clone my fork of TheiaSfM?  This function is included in this fork https://github.com/urbste/TheiaSfM/blob/feature/r7pfr/src/theia/sfm/estimators/estimate_radial_dist_uncalibrated_absolute_pose.h

git clone https://github.com/urbste/TheiaSfM
cd TheiaSfM && git checkout cda5d4f && mkdir -p build && cd build
cmake .. && make -j4
sudo make install

gongping108, 2021-06-24:
Thanks! I always use your TheiaSfm! finally error: double free or corruption (out) In estimator.h
    std::vector<Model> temp_models;
    if (!estimator_.EstimateModel(data_subset, &temp_models)) {
      continue;
    }

gongping108, 2021-06-28:
I make TheiaSfM, option(BUILD_TESTING "Enable testing" ON) ! All tests passed!

gongping108, 2021-06-30:
Thanks urbste! After updated The problem  has been solved !

closed 2021-06-30 by gongping108