urbste/OpenImuCameraCalibrator #7: static_calib_result.json

gongping108, 2021-06-30:
Can you provide static_calib_result.json or dataset?

gongping108, 2021-06-30:
I don't have the static_calib_result.json.
 I  comment "--imu_intrinsics="+args.path_to_imu_intrinsics, 
In  double reproj_error =
      imu_cam_calibrator.Optimize(20, false, false, false, true);  
The following error:
*** stack smashing detected ***: <unknown> terminated
the error occurrs in meanRSReprojection

urbste, 2021-08-15:
Sorry for my late reply. I added a [description](https://github.com/urbste/OpenImuCameraCalibrator#example-intrinsic-imu-parameter-estimation)  on how to generate this file. I also updated the dataset you can find on Google Drive. It contains a sample file.

gongping108, 2021-08-17:
Thank you very much!

closed 2021-08-17 by gongping108