urbste/OpenImuCameraCalibrator #8: Error with tbb library

biendltb, 2021-08-14:
Thanks for the great work.

When I was building the project, there are some packages I think you missed in README:
1. `opencv_contrib` should be built together with OpenCV since you use aruco APIs in your code which is in the `opencv_contrib` package. It should be stated explicitly in the installation instruction.
2. In the installation, `libtbb-dev` is required but it didn't mention. I installed the library from APT but it didn't work.

The `libtbb-dev` causes an error as the log below. Do you know how to fix it? Thanks in advance.

```txt
biendltb@bienpc:/data/Projects/tmp/OpenImuCameraCalibrator/build$ make -j15
[  2%] Building CXX object CMakeFiles/OpenImuCameraCalibrator.dir/src/core/imu_camera_calibrator.cc.o
In file included from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/calib_helpers.h:38,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/imu_camera_calibrator.h:22,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/src/core/imu_camera_calibrator.cc:16:
/data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/common_types.h:129:10: warning: ‘template<class Key> class tbb::tbb_hash’ is deprecated: tbb::tbb_hash is deprecated, use std::hash [-Wdeprecated-declarations]
  129 |     tbb::tbb_hash<std::pair<TimeCamId, TimeCamId>>,
      |          ^~~~~~~~
In file included from /usr/include/tbb/internal/_concurrent_unordered_impl.h:48,
                 from /usr/include/tbb/concurrent_unordered_map.h:26,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/common_types.h:41,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/calib_helpers.h:38,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/imu_camera_calibrator.h:22,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/src/core/imu_camera_calibrator.cc:16:
/usr/include/tbb/internal/_tbb_hash_compare_impl.h:86:74: note: declared here
   86 | class __TBB_DEPRECATED_MSG("tbb::tbb_hash is deprecated, use std::hash") tbb_hash
      |                                                                          ^~~~~~~~
In file included from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/calib_helpers.h:38,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/imu_camera_calibrator.h:22,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/src/core/imu_camera_calibrator.cc:16:
/data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/common_types.h:304:8: warning: ‘template<class Key> class tbb::tbb_hash’ is deprecated: tbb::tbb_hash is deprecated, use std::hash [-Wdeprecated-declarations]
  304 | struct tbb_hash<TimeCamId> : public std::hash<TimeCamId> {};
      |        ^~~~~~~~
In file included from /usr/include/tbb/internal/_concurrent_unordered_impl.h:48,
                 from /usr/include/tbb/concurrent_unordered_map.h:26,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/common_types.h:41,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/calib_helpers.h:38,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/imu_camera_calibrator.h:22,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/src/core/imu_camera_calibrator.cc:16:
/usr/include/tbb/internal/_tbb_hash_compare_impl.h:86:74: note: declared here
   86 | class __TBB_DEPRECATED_MSG("tbb::tbb_hash is deprecated, use std::hash") tbb_hash
      |                                                                          ^~~~~~~~
In file included from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/calib_helpers.h:38,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/imu_camera_calibrator.h:22,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/src/core/imu_camera_calibrator.cc:16:
/data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/common_types.h:307:8: warning: ‘template<class Key> class tbb::tbb_hash’ is deprecated: tbb::tbb_hash is deprecated, use std::hash [-Wdeprecated-declarations]
  307 | struct tbb_hash<std::pair<TimeCamId, TimeCamId>>
      |        ^~~~~~~~
In file included from /usr/include/tbb/internal/_concurrent_unordered_impl.h:48,
                 from /usr/include/tbb/concurrent_unordered_map.h:26,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/common_types.h:41,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/calib_helpers.h:38,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/imu_camera_calibrator.h:22,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/src/core/imu_camera_calibrator.cc:16:
/usr/include/tbb/internal/_tbb_hash_compare_impl.h:86:74: note: declared here
   86 | class __TBB_DEPRECATED_MSG("tbb::tbb_hash is deprecated, use std::hash") tbb_hash
      |                                                                          ^~~~~~~~
In file included from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/basalt_spline/calib_helpers.h:41,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/imu_camera_calibrator.h:22,
                 from /data/Projects/tmp/OpenImuCameraCalibrator/src/core/imu_camera_calibrator.cc:16:
/usr/include/tbb/tbb.h:21:154: note: #pragma message: TBB Warning: tbb.h contains deprecated functionality. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | contains deprecated functionality. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                      ^

[  5%] Linking CXX static library libOpenImuCameraCalibrator.a
[ 56%] Built target OpenImuCameraCalibrator
Scanning dependencies of target estimate_imu_to_camera_rotation
Scanning dependencies of target static_imu_calibration
Scanning dependencies of target extract_board_to_json
Scanning dependencies of target fit_allan_variance
Scanning dependencies of target estimate_camera_poses_from_checkerboard
[... 83 lines cut ...]
[ 97%] Linking CXX executable static_imu_calibration
[ 97%] Built target estimate_camera_poses_from_checkerboard
[ 97%] Built target static_imu_calibration
[ 97%] Built target calibrate_camera
[100%] Linking CXX executable continuous_time_imu_to_camera_calibration
/usr/bin/ld: CMakeFiles/continuous_time_imu_to_camera_calibration.dir/continuous_time_imu_to_camera_calibration.cc.o: in function `tbb::interface7::task_arena::initialize()':
continuous_time_imu_to_camera_calibration.cc:(.text._ZN3tbb10interface710task_arena10initializeEv[_ZN3tbb10interface710task_arena10initializeEv]+0x27): undefined reference to `tbb::interface7::internal::task_arena_base::internal_initialize()'
/usr/bin/ld: CMakeFiles/continuous_time_imu_to_camera_calibration.dir/continuous_time_imu_to_camera_calibration.cc.o: in function `tbb::interface7::task_arena::terminate()':
continuous_time_imu_to_camera_calibration.cc:(.text._ZN3tbb10interface710task_arena9terminateEv[_ZN3tbb10interface710task_arena9terminateEv]+0x24): undefined reference to `tbb::interface7::internal::task_arena_base::internal_terminate()'
/usr/bin/ld: CMakeFiles/continuous_time_imu_to_camera_calibration.dir/continuous_time_imu_to_camera_calibration.cc.o: in function `tbb::interface7::task_arena::current_thread_index()':
continuous_time_imu_to_camera_calibration.cc:(.text._ZN3tbb10interface710task_arena20current_thread_indexEv[_ZN3tbb10interface710task_arena20current_thread_indexEv]+0x9): undefined reference to `tbb::interface7::internal::task_arena_base::internal_current_slot()'
/usr/bin/ld: CMakeFiles/continuous_time_imu_to_camera_calibration.dir/continuous_time_imu_to_camera_calibration.cc.o: in function `tbb::flow::interface10::graph::wait_for_all()':
[... cut ...] ined reference to `tbb::interface7::internal::task_arena_base::internal_execute(tbb::interface7::internal::delegate_base&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [applications/CMakeFiles/continuous_time_imu_to_camera_calibration.dir/build.make:125: applications/continuous_time_imu_to_camera_calibration] Error 1
make[1]: *** [CMakeFiles/Makefile2:191: applications/CMakeFiles/continuous_time_imu_to_camera_calibration.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

urbste, 2021-08-15:
Thanks for the hint I updated the readme.

I removed all TBB includes, as I do not use the respective basalt types anyways. Could you please pull and retry?

biendltb, 2021-08-15:
Hi Steffen,

Thanks for the fast response.

The build errors that I encountered disappear now but some more errors appear when I try to run the program:
1) The program requires `--path_to_build` parameters but it was not stated in your sample command. It is an easy fix to add it but I think you could point it to $PWD/build/applications as the default path instead of the hard-coded path of your machine. Otherwise, you could state it explicitly in your sample command. A similar fix could be applied to `--path_to_src`.

2) I downloaded your dataset3 from the Google Drive URL that you provided but it doesn't contain the IMU data. Specifically, the program tries to look for the file `dataset3/cam_imu/GH010164.json`. Could you update your sample dataset with the IMU data also?

Thank you very much for your help.

Bien

urbste, 2021-08-15:
The IMU data should be extracted by the script from the MP4 file (line 176 in run_gopro_*.py). You can check if your node installation is working correctly by running:
```bash
cd javascript
node extract_metadata.js /path/to/sample/folder GH010164.MP4 /path/to/sample/folder
```

urbste, 2021-08-15:
I also updated the documentation on how to do the static imu calibration if you want to try that out. The example file now also contains a MP4 to do that. You just need to re-download the GoPro9.zip file.

biendltb, 2021-08-15:
Hi Steffen, thanks for the reply. I didn't trace up the error log so I just noticed that the nodejs requires the module `mp4box`. I didn't read the README inside the javascript folder which shows that I should run `npm install` to have all the required modules installed. I'm not familiar with nodejs so I didn't know it. I think it should be placed in the main readme after installing nodejs, too.

Thanks for updating the documentation with the static IMU calibration, I sure will try it out.

After fixing the problem with extracting metadata, there is another issue that the script looks for the file `dataset3/cam/cam_calib_GH010162_di_2.0.json`. I don't know this file is generated in which script. I could see this file in your updated dataset but I just don't want to miss the step that generated the file.

Full log of the error:
```bash
biendltb@bienpc:/data/Projects/tmp/OpenImuCameraCalibrator$ python python/run_gopro_calibration.py --path_calib_dataset=/home/biendltb/Downloads/GoPro9/dataset3/ --path_to_build=/data/Projects/tmp/OpenImuCameraCalibrator/build/applications --path_to_src=/data/Projects/tmp/OpenImuCameraCalibrator/ --checker_size_m=0.021 --image_downsample_factor=2 --camera_model=DIVISION_UNDISTORTION
['/home/biendltb/Downloads/GoPro9/dataset3/cam/GH010162.MP4']
==================================================================
Running corner extraction.
==================================================================
Extracing corners for camera calibration.
I0815 18:07:33.267990 98400 extract_board_to_json.cc:82] Starting board extraction. This might take a while...
Total number of frames: 0
Extracing corners for imu camera calibration.
I0815 18:07:33.271963 98401 extract_board_to_json.cc:82] Starting board extraction. This might take a while...
Total number of frames: 0
Finished corner extraction.
==================================================================
Corner extraction took 0.01s.
==================================================================
==================================================================
Running camera calibration.
==================================================================
Calibrating camera.
calibrate_camera: /data/Projects/tmp/OpenImuCameraCalibrator/include/OpenCameraCalibrator/utils/json.h:17760: const value_type& nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>::operator[](T*) const [with T = const char; ObjectType = std::map; ArrayType = std::vector; StringType = std::__cxx11::basic_string<char>; BooleanType = bool; NumberIntegerType = long int; NumberUnsignedType = long unsigned int; NumberFloatType = double; AllocatorType = std::allocator; JSONSerializer = nlohmann::adl_serializer; nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>::const_reference = const nlohmann::basic_json<>&; nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>::value_type = nlohmann::basic_json<>]: Assertion `m_value.object->find(key) != m_value.object->end()' failed.
Finished camera calibration.
==================================================================
Camera calibration took 0.07s.
==================================================================
==================================================================
Extracting GoPro telemetry for imu bias and camera imu calibration.
==================================================================
Input video path: /home/biendltb/Downloads/GoPro9/dataset3/cam_imu/GH010164.MP4
Output file will be: /home/biendltb/Downloads/GoPro9/dataset3/cam_imu/GH010164.MP4_telemetry.json
Opening /home/biendltb/Downloads/GoPro9/dataset3/cam_imu/GH010164.MP4
Input video path: /home/biendltb/Downloads/GoPro9/dataset3/imu_bias/GH010163.MP4
Output file will be: /home/biendltb/Downloads/GoPro9/dataset3/imu_bias/GH010163.MP4_telemetry.json
Opening /home/biendltb/Downloads/GoPro9/dataset3/imu_bias/GH010163.MP4
Read stream data from GH010163.MP4
Extracted information for GH010163.MP4
Flushing on end after 67141604 bytes
Generated telemetry for GH010163.MP4
Telemetry saved as JSON to GH010163.json
Done
Read stream data from GH010164.MP4
Extracted information for GH010164.MP4
Flushing on end after 440372248 bytes
Generated telemetry for GH010164.MP4
Telemetry saved as JSON to GH010164.json
Done
==================================================================
Telemetry extraction took 0.64s.
==================================================================
==================================================================
Estimating IMU biases.
==================================================================
Estimated biases:
gyroscope bias:     -0.00161 rad/s, 0.00187 rad/s, -0.00759 rad/s
accelerometer bias: -0.02881 m/s2,  -0.03653 m/s2,  0.01007 m/s2
Writing result to:  /home/biendltb/Downloads/GoPro9/dataset3/imu_bias/imu_bias_GH010163.json
==================================================================
IMU bias estimation took 0.27s.
==================================================================
==================================================================
Estimating camera poses for IMU - CAM calibration.
==================================================================
[... cut ...]
  File "/data/Projects/tmp/OpenImuCameraCalibrator/python/print_result_stats.py", line 75, in <module>
    main()
  File "/data/Projects/tmp/OpenImuCameraCalibrator/python/print_result_stats.py", line 24, in main
    data = read_calib_json(args.path_results)
  File "/data/Projects/tmp/OpenImuCameraCalibrator/python/print_result_stats.py", line 9, in read_calib_json
    with open(file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/biendltb/Downloads/GoPro9/dataset3/cam_imu/cam_imu_calib_result_GH010164.json'
```

biendltb, 2021-08-16:
Hi Steffen,

I think the above error is caused by the corner extractor. I try to separate it from the pipeline by executing the command:

```bash
biendltb@bienpc:OpenImuCameraCalibrator (master)$ ./build/applications/extract_board_to_json --input_path=/home/biendltb/Downloads/GoPro9/dataset3/cam/GH010162.MP4 --aruco_detector_params=/data/Projects/tmp/OpenImuCameraCalibrator/resource/charuco_detector_params.yml --board_type=charuco --save_corners_json_path=/home/biendltb/Downloads/GoPro9/dataset3/cam/cam_corners_GH010162.uson --downsample_factor=2.0 --checker_square_length_m=0.021 --verbose=0 --recompute_corners=0 --num_squares_x=10 --num_squares_y=8 --logtostderr=1
```

And got the result:

```txt
I0816 19:28:49.831037 85041 extract_board_to_json.cc:82] Starting board extraction. This might take a while...
Total number of frames: 0
```

So the charuco board detector cannot recognize the pattern in the video. So I got a 2.2KB binary file for the `cam_corners_GH010162.uson` which I think it's empty.

urbste, 2021-08-16:
I just tried the dataset myself and the extraction ist working... 
Total number of frames: 0 basically means that OpenCV is not able to open the video stream.

Maybe you are missing some codec or something when compiling OpenCV?

biendltb, 2021-08-17:
Hi Steffen,

Thanks for the hint. It's a stupid mistake when I didn't notice that the FFMPEG wasn't set up correctly when I built OpenCV. I was already able to run your script with static imu calibration and the visual-inertial calibration. Now I will have to analyse the data and see how should I apply the same technique on my own dataset. Thanks again for the great work, I think the ticket can be closed here.

closed 2021-08-18 by biendltb