urbste/OpenImuCameraCalibrator #10: Remove SSE2 on ARM64

SKioresku, 2021-10-05:
I am trying to compile without SSE2 but I get errors.

urbste, 2021-10-05:
Hi, first you would need to be a bit more specific: Where does the error occur? Is it in the OpenICC or another library?

Second I never tried to compile on ARM. Why would you want to do that?

SKioresku, 2021-10-05:
Hi. I am building a library on nvidia jetson nano B1 to use it as a base for a robot

SKioresku, 2021-10-05:
An error occurred in the vlfeat library

SKioresku, 2021-10-05:
I want to transfer data from android to Nvidia Jetson to calibrate sensors

urbste, 2021-10-05:
Ok. Could you not calibrate the sensors on a Laptop or a desktop PC?
 
I do not have the capacity to update any third party libraries sorry :/

SKioresku, 2021-10-05:
I need to have the opportunity for the one who buys the robot to calibrate with it ))) I will try to disable SSE2 myself. It seems to be possible there, I just thought that TheiaSFM should have an option. Thanks))

closed 2021-10-05 by SKioresku