urbste/OpenImuCameraCalibrator #12: error: ‘PnPType’ is not a member of ‘theia’

wolf943134497, 2022-01-20:
Hi @urbste  Thanks for your works!

An error occurred when use the new version of OpenICC and pyTheiaSFM .
/OpenImuCameraCalibrator/include/OpenCameraCalibrator/core/pose_estimator.h:80:10: error: ‘PnPType’ in namespace ‘theia’ does not name a type
   theia::PnPType pnp_type_ = theia::PnPType::DLS;

I noticed that you are using the new pnp method,  when I use last version, The gopro calibration process seems to be stuck,  it seems that there is a problem with pose  optimization.
I20220119 21:06:09.460232 137055 pose_estimator.cc:132] Skipping view at timestamp : 55.8892s. Not enough points found.
I20220119 21:06:09.460340 137055 pose_estimator.cc:132] Skipping view at timestamp : 55.9059s. Not enough points found.
I20220119 21:06:09.510980 137055 pose_estimator.cc:132] Skipping view at timestamp : 56.69s. Not enough points found.
I20220119 21:06:09.514223 137055 pose_estimator.cc:132] Skipping view at timestamp : 56.7233s. Not enough points found.
I20220119 21:06:09.541810 137055 pose_estimator.cc:132] Skipping view at timestamp : 57.2906s. Not enough points found.
I20220119 21:06:09.562834 137055 pose_estimator.cc:132] Skipping view at timestamp : 57.8411s. Not enough points found.
I20220119 21:06:09.570690 137055 pose_estimator.cc:132] Skipping view at timestamp : 58.0914s. Not enough points found.
I20220119 21:06:09.600827 137055 pose_estimator.cc:146] Pose estimation failed for view at timestamp 58.6085s.
I20220119 21:06:09.702355 137055 pose_estimator.cc:132] Skipping view at timestamp : 60.0934s. Not enough points found.
I20220119 21:06:09.741844 137055 pose_estimator.cc:132] Skipping view at timestamp : 60.744s. Not enough points found.
I20220119 21:06:09.742926 137055 pose_estimator.cc:132] Skipping view at timestamp : 60.7774s. Not enough points found.
I20220119 21:06:09.742964 137055 pose_estimator.cc:132] Skipping view at timestamp : 60.7941s. Not enough points found.
I20220119 21:06:09.744585 137055 pose_estimator.cc:132] Skipping view at timestamp : 60.9108s. Not enough points found.
I20220119 21:06:09.778606 137055 pose_estimator.cc:132] Skipping view at timestamp : 61.3613s. Not enough points found.
I20220119 21:06:09.783015 137055 pose_estimator.cc:132] Skipping view at timestamp : 61.428s. Not enough points found.
I20220119 21:06:09.783051 137055 pose_estimator.cc:132] Skipping view at timestamp : 61.4447s. Not enough points found.
I20220119 21:06:09.783080 137055 pose_estimator.cc:132] Skipping view at timestamp : 61.4781s. Not enough points found.
I20220119 21:06:09.813349 137055 pose_estimator.cc:132] Skipping view at timestamp : 61.8284s. Not enough points found.
I20220119 21:06:09.843387 137055 pose_estimator.cc:132] Skipping view at timestamp : 62.4123s. Not enough points found.
I20220119 21:06:09.843415 137055 pose_estimator.cc:132] Skipping view at timestamp : 62.5124s. Not enough points found.
I20220119 21:06:09.876546 137055 pose_estimator.cc:132] Skipping view at timestamp : 63.0129s. Not enough points found.
I20220119 21:06:09.878156 137055 pose_estimator.cc:132] Skipping view at timestamp : 63.0797s. Not enough points found.
I20220119 21:06:09.878191 137055 pose_estimator.cc:132] Skipping view at timestamp : 63.0964s. Not enough points found.
I20220119 21:06:09.878226 137055 pose_estimator.cc:132] Skipping view at timestamp : 63.1297s. Not enough points found.
I20220119 21:06:09.897560 137055 pose_estimator.cc:132] Skipping view at timestamp : 63.3466s. Not enough points found.
I20220119 21:06:09.897598 137055 pose_estimator.cc:132] Skipping view at timestamp : 63.3633s. Not enough points found.
I20220119 21:06:09.900799 137055 pose_estimator.cc:132] Skipping view at timestamp : 63.4634s. Not enough points found.
I20220119 21:06:09.962488 137055 pose_estimator.cc:132] Skipping view at timestamp : 64.3643s. Not enough points found.
I20220119 21:06:09.963955 137055 pose_estimator.cc:132] Skipping view at timestamp : 64.4477s. Not enough points found.
I20220119 21:06:09.963982 137055 pose_estimator.cc:132] Skipping view at timestamp : 64.5144s. Not enough points found.
I20220119 21:06:10.001494 137055 pose_estimator.cc:132] Skipping view at timestamp : 64.8982s. Not enough points found.
I20220119 21:06:10.001520 137055 pose_estimator.cc:132] Skipping view at timestamp : 64.9482s. Not enough points found.
I20220119 21:06:10.001549 137055 pose_estimator.cc:132] Skipping view at timestamp : 64.9649s. Not enough points found.
I20220119 21:06:10.004623 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.115s. Not enough points found.
I20220119 21:06:10.004660 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.1317s. Not enough points found.
I20220119 21:06:10.025790 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.3486s. Not enough points found.
I20220119 21:06:10.027387 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.432s. Not enough points found.
I20220119 21:06:10.030544 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.4988s. Not enough points found.
I20220119 21:06:10.031658 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.5321s. Not enough points found.
I20220119 21:06:10.033329 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.5822s. Not enough points found.
I20220119 21:06:10.058063 137055 pose_estimator.cc:132] Skipping view at timestamp : 65.8825s. Not enough points found.
I20220119 21:06:10.060990 137055 pose_estimator.cc:132] Skipping view at timestamp : 66.0994s. Not enough points found.
I20220119 21:06:10.081813 137055 pose_estimator.cc:132] Skipping view at timestamp : 66.3496s. Not enough points found.
I20220119 21:06:10.113782 137055 pose_estimator.cc:132] Skipping view at timestamp : 66.8501s. Not enough points found.
I20220119 21:06:11.379074 137055 estimate_camera_poses_from_checkerboard.cc:58] Finished pose estimation.

urbste, 2022-01-20:
I just pushed a major update. Please pull and try again :)

wolf943134497, 2022-01-20:
Thanks for your quick reply and suggestion.

closed 2022-01-20 by wolf943134497