urbste/OpenImuCameraCalibrator #16: question about the concatenation of metadata

CanCanZeng, 2022-04-02:
Hi @urbste , thanks for sharing this great work!
I'm also trying to use the video file recorded by GoPro to run vio. I'm so happy when finding someone already done so much research!

My question is about this line https://github.com/urbste/OpenImuCameraCalibrator/blob/b7ba9940638e5a538b421dbaeaf3717f8560d512/python/telemetry_converter.py#L48
Is it reasonable to just concatenate the metadata directly? Shouldn't there be some time gap between adjacent video segments?

urbste, 2022-04-04:
Hi, I never though about it. I was actually thinking, that the GoPro just splits the entire stream into multiple sub MP4s if it get's too large to save it. So why do you think there should be a time gap between them?

CanCanZeng, 2022-04-18:
Sorry, I found out that I had this problem because I used someone else's code. He converts the absolute time in each video into relative time relative to the beginning of each video.
Thanks for opening this project again, it helps me a lot!

closed 2022-04-27 by CanCanZeng