urbste/OpenImuCameraCalibrator #22: Failed in pose estimation

yjjuan, 2022-09-03:
Hi urbste,
I have met error `std::bad_alloc` during pose optimization. Do you have any idea about this error?  Thanks in advance. 
```log
Track Id: 55 std dev: 0.0409686 0.0265254 0.0545904 mm
Track Id: 56 std dev: 0.0385919 0.0249822 0.0543944 mm
Track Id: 57 std dev: 0.0358216 0.0243497 0.0538741 mm
Track Id: 58 std dev: 0.0335099 0.0238545  0.053015 mm
Track Id: 59 std dev: 0.0322875 0.0234659 0.0525263 mm
Track Id: 60 std dev: 0.0322966 0.0234475 0.0527493 mm
Track Id: 61 std dev: 0.0328023 0.0236162 0.0530119 mm
Track Id: 62 std dev:  0.034797 0.0250347 0.0553444 mm
Mean board point standard deviation after optimization: 0.0281578 0.0246573 0.0531155 mm
I0903 05:02:06.131096 249378 pose_estimator.cc:231] Optimizing all estimated poses.
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
==================================================================
Pose estimation estimation took 43.33s.
```

urbste, 2022-11-06:
Hi, sorry for the late reply. I have the same issue. I will try to investigate this. You can try to set the flag --optimize_board_points=0 It seems there is a problem with BundleAdjustViews in pyTheiaSfM after board optimization.

urbste, 2022-11-06:
Fixed the bug.

closed 2022-11-06 by urbste