urbste/OpenImuCameraCalibrator #23: Issues running your example with GoPro9

franillo, 2022-11-23:
Hi, I am trying to follow your calibration example with your data as explained here:
[https://github.com/urbste/OpenImuCameraCalibrator/blob/master/docs/gopro_calibration.md](url)

If I extract your data and delete all the files except the mp4 files, I get following error:

==================================================================
Optimizing IMU to Camera calibration using Spline Fusion.
==================================================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1123 11:35:26.019022 925709 continuous_time_imu_to_camera_calibration.cc:167] Check failed: ReadIMU2CamInit( FLAGS_gyro_to_cam_initial_calibration, imu2cam, time_offset_imu_to_cam) Could not read: /home/francisco/Downloads/GoPro9/dataset3/cam_imu/imu_to_cam_calibration_GH010164.json

If I extract the files provided and do not delete the calibration files, the program runs through with no problems.
I hope somebody can help, this error might be related to some issues already mentioned here with the pyTheia lib?
I have also a few of these here:

I1123 11:55:45.336704 926842 pose_estimator.cc:254] Removing view 2022 due to large z coordinate: -0.387445 vs median z coordinate -0.183183

urbste, 2023-02-24:
Sorry for my late reply. I cannot reproduce these errors for the example dataset. Are you sure that you have all the latest versions of all libraries built?
I added a Dockerfile and a [description](https://github.com/urbste/OpenImuCameraCalibrator#dockerfile) how to use it. Maybe check that all the versions are correct.

urbste, 2023-02-24:
I also update the GoPro9.zip file and removed all temporary files. Maybe there were some temp files responsible for these errors.