urbste/OpenImuCameraCalibrator #29: Various files missing when calibrating.

ajsnyde, 2023-02-25:
[image]

The corners are extracted, lots of stuff is generated, but I see quite a few errors towards the end.

Command within the docker container is: ` python3 python/run_gopro_calibration.py --path_calib_dataset ../dataset/calibrationDataset/ --checker_size_m=0.05115 --path_to_build ../OpenImuCameraCalibrator/build/applications/`

Command to start up the docker container: `sudo docker run -it --rm -v `pwd`:/home -v /home/Downloads/GoPro9:/dataset openicc`

ajsnyde, 2023-02-25:
The MP4s are from a Hero11, for what that's worth.

urbste, 2023-03-02:
Hi, can you share your dataset so I can have a look?

anaidasdfg, 2023-03-02:
Same problem here! I tried it with the given videos for [GoPro 9 examples](https://github.com/urbste/OpenImuCameraCalibrator/blob/master/docs/gopro_calibration.md) and had the same issue.

ajsnyde, 2023-03-05:
https://user-images.githubusercontent.com/11302823/222992366-b260e63d-2044-4e70-bd1c-d87c093187bf.mp4

Here's an mp4 I'm trying to calibrate with. I know, not ideal calibration material, but this is the one under 10mb.

urbste, 2023-03-06:
@anaidasdfg Sorry, there was the Dockerfile was missing an npm install in the javascript folder. I pushed an update please check it out :)

urbste, 2023-03-06:
@ajsnyde Is that the original file from your GoPro? It does not seem to contain a metadata stream:

I am trying to extract the metadata with the script **extract_telemetry_from_folder.py*
`
python python/extract_telemetry_from_folder.py --path ../dataset/GoProAjsnyde
`
But get the following error:

`Read stream data from 222992366-b260e63d-2044-4e70-bd1c-d87c093187bf.MP4
An error occurred Track not found
Flushing on end after 10345190 bytes`

ajsnyde, 2023-03-06:
Actually, I screwed up, that's not the file I was using to calibrate the IMU. I've got a dataset but it's like 2GB in size, and can't upload it

ajsnyde, 2023-03-06:
```root@7ce3ee00fab5:/OpenImuCameraCalibrator# python3 python/run_gopro_calibration.py --path_calib_dataset ../dataset/calibrationDataset/ --checker_size_m=0.05115 --path_to_build ../OpenImuCameraCalibrator/build/applications/
Error! Could not find cam calibration video file with MP4 ending in path ../dataset/calibrationDataset/cam
root@7ce3ee00fab5:/OpenImuCameraCalibrator# python3 python/run_gopro_calibration.py --path_calib_dataset ../dataset/ --checker_size_m=0.05115 --path_to_build ../OpenImuCameraCalibrator/build/applications/
['../dataset/cam/GX010027.MP4']
==================================================================
Running corner extraction.
==================================================================
Extracing corners for camera calibration.
I0306 14:04:39.909039    78 extract_board_to_json.cc:92] Starting board extraction. This might take a while...
Total number of frames: 4062
I0306 14:04:41.944630    78 board_extractor.cc:443] Extracting corners from frame 60 / 4062
I0306 14:04:43.780072    78 board_extractor.cc:443] Extracting corners from frame 120 / 4062
I0306 14:04:45.552951    78 board_extractor.cc:443] Extracting corners from frame 180 / 4062
I0306 14:04:47.428068    78 board_extractor.cc:443] Extracting corners from frame 240 / 4062
I0306 14:04:49.350881    78 board_extractor.cc:443] Extracting corners from frame 300 / 4062
I0306 14:04:51.299734    78 board_extractor.cc:443] Extracting corners from frame 360 / 4062
I0306 14:04:53.173480    78 board_extractor.cc:443] Extracting corners from frame 420 / 4062
I0306 14:04:55.114609    78 board_extractor.cc:443] Extracting corners from frame 480 / 4062
I0306 14:04:57.007761    78 board_extractor.cc:443] Extracting corners from frame 540 / 4062
I0306 14:04:58.913098    78 board_extractor.cc:443] Extracting corners from frame 600 / 4062
I0306 14:05:01.036238    78 board_extractor.cc:443] Extracting corners from frame 660 / 4062
I0306 14:05:03.129483    78 board_extractor.cc:443] Extracting corners from frame 720 / 4062
I0306 14:05:04.913065    78 board_extractor.cc:443] Extracting corners from frame 780 / 4062
I0306 14:05:06.676331    78 board_extractor.cc:443] Extracting corners from frame 840 / 4062
I0306 14:05:08.450070    78 board_extractor.cc:443] Extracting corners from frame 900 / 4062
I0306 14:05:10.230935    78 board_extractor.cc:443] Extracting corners from frame 960 / 4062
I0306 14:05:12.069403    78 board_extractor.cc:443] Extracting corners from frame 1020 / 4062
I0306 14:05:13.939038    78 board_extractor.cc:443] Extracting corners from frame 1080 / 4062
I0306 14:05:15.853615    78 board_extractor.cc:443] Extracting corners from frame 1140 / 4062
I0306 14:05:17.852705    78 board_extractor.cc:443] Extracting corners from frame 1200 / 4062
I0306 14:05:19.950106    78 board_extractor.cc:443] Extracting corners from frame 1260 / 4062
I0306 14:05:22.104259    78 board_extractor.cc:443] Extracting corners from frame 1320 / 4062
I0306 14:05:24.284386    78 board_extractor.cc:443] Extracting corners from frame 1380 / 4062
I0306 14:05:26.340768    78 board_extractor.cc:443] Extracting corners from frame 1440 / 4062
I0306 14:05:28.352627    78 board_extractor.cc:443] Extracting corners from frame 1500 / 4062
I0306 14:05:30.375968    78 board_extractor.cc:443] Extracting corners from frame 1560 / 4062
I0306 14:05:32.285068    78 board_extractor.cc:443] Extracting corners from frame 1620 / 4062
I0306 14:05:34.274109    78 board_extractor.cc:443] Extracting corners from frame 1680 / 4062
I0306 14:05:36.254408    78 board_extractor.cc:443] Extracting corners from frame 1740 / 4062
I0306 14:05:38.407099    78 board_extractor.cc:443] Extracting corners from frame 1800 / 4062
I0306 14:05:40.439405    78 board_extractor.cc:443] Extracting corners from frame 1860 / 4062
I0306 14:05:42.557495    78 board_extractor.cc:443] Extracting corners from frame 1920 / 4062
I0306 14:05:44.681175    78 board_extractor.cc:443] Extracting corners from frame 1980 / 4062
I0306 14:05:46.752436    78 board_extractor.cc:443] Extracting corners from frame 2040 / 4062
I0306 14:05:48.889516    78 board_extractor.cc:443] Extracting corners from frame 2100 / 4062
I0306 14:05:50.976758    78 board_extractor.cc:443] Extracting corners from frame 2160 / 4062
I0306 14:05:53.016901    78 board_extractor.cc:443] Extracting corners from frame 2220 / 4062
I0306 14:05:55.096585    78 board_extractor.cc:443] Extracting corners from frame 2280 / 4062
I0306 14:05:57.129492    78 board_extractor.cc:443] Extracting corners from frame 2340 / 4062
I0306 14:05:59.254082    78 board_extractor.cc:443] Extracting corners from frame 2400 / 4062
I0306 14:06:01.556769    78 board_extractor.cc:443] Extracting corners from frame 2460 / 4062
I0306 14:06:03.716398    78 board_extractor.cc:443] Extracting corners from frame 2520 / 4062
I0306 14:06:05.945624    78 board_extractor.cc:443] Extracting corners from frame 2580 / 4062
I0306 14:06:07.992089    78 board_extractor.cc:443] Extracting corners from frame 2640 / 4062
I0306 14:06:09.947945    78 board_extractor.cc:443] Extracting corners from frame 2700 / 4062
I0306 14:06:12.069813    78 board_extractor.cc:443] Extracting corners from frame 2760 / 4062
I0306 14:06:14.257180    78 board_extractor.cc:443] Extracting corners from frame 2820 / 4062
I0306 14:06:16.562214    78 board_extractor.cc:443] Extracting corners from frame 2880 / 4062
[... cut ...]
==================================================================
Print results.
==================================================================
Traceback (most recent call last):
  File "/OpenImuCameraCalibrator/python/../python/print_result_stats.py", line 98, in <module>
    main()
  File "/OpenImuCameraCalibrator/python/../python/print_result_stats.py", line 25, in main
    data = dict(data["trajectory"]) 
KeyError: 'trajectory'
root@7ce3ee00fab5:/OpenImuCameraCalibrator# 
```
Here's the latest run, with the latest built docker container

ajsnyde, 2023-03-06:
> Btw when I see your underwater imagery, I am not sure if you can directly use the toolbox without modification. One probably needs to adjust some stuff according to that [paper](https://openaccess.thecvf.com/content_cvpr_2018_workshops/papers/w28/Pedersen_Camera_Calibration_for_CVPR_2018_paper.pdf)

I could be totally wrong here, but I interpreted the paper as looking to calibrate a camera or two _above_ water and using some math to adapt the calibration for underwater usage. I'm looking to calibrate the camera underwater from the start (despite the dataset I'm uploading, that's to POC the IMU integration). Assuming you don't change mediums, the surrounding water is just another lens as far as SLAM is concerned.

Then again, I could be totally wrong, so if I am, let me know :smile:

urbste, 2023-03-07:
I had a look at your sample dataset. The issue is, that you are using a standard checkerboard. This is not supported by my library. We need to make the orientation of the board w.r.t to the camera fix, to be able to align the coordinate systems of the board and the world (gravity) correctly. 

Please use a the Charuco board: https://github.com/urbste/OpenImuCameraCalibrator/blob/master/resource/board.png

ajsnyde, 2023-03-08:
Thanks! I've redone the calibration videos using your board, and now it's hanging:
[image]
Link to the new calibration dataset. Running command `python3 python/run_gopro_calibration.py --path_calib_dataset ../dataset --checker_size_m=0.05115 --path_to_build ../OpenImuCameraCalibrator/build/applications/ --camera_model=FISHEYE`
https://drive.google.com/file/d/1WwSFUW4v7m_MphEzr3MWnLAJbQKUrnvU/view?usp=sharing

ajsnyde, 2023-03-08:
Also trying downsized videos (like 30 mb instead of 500mb) to see if it's related to the resolution, but it's also hanging for 20+ minutes.

urbste, 2023-03-08:
I just --voxel_grid_size=0.06 and I get (with --image_downsample_factor=4):

`Using 68 views for camera calibration.
Final camera calibration reprojection error: 0.721298 from 47 view.
Focal Length:425.1px Principal Point: 490.029/417.294px.
FISHEYE model: Radial distortion 1: 0.0428467 Radial distortion 2: 0.212038 Radial distortion 3: -0.334151 Radial distortion 4: 0.0767752
Finished camera calibration.`

urbste, 2023-03-08:
What type of resolution setting is that? I have never seen 3840x3360?

ajsnyde, 2023-03-08:
That's the latest GoPro Hero 11, with 4k 8:7 @60fps. No image stabilization (which tends to crop the video a bit).

urbste, 2023-03-08:
Ah cool. Sure, for IMU-Camera calibration this needs to turned off anyways ;)

urbste, 2023-03-08:
Ok now I am getting a weird error in the rotation initialization. I will check what's wrong there.

urbste, 2023-03-08:
The camera poses for visual-inertial calibration look alright, maybe not enough motion
[image]

ajsnyde, 2023-03-08:
I'll make a longer recording for the cam-imu data. Should I try to always keep the board taking up most of the video, or is it ok to back away from it a little?

urbste, 2023-03-08:
It does not need to be in the entire view. Actually being a bit closer is better, The further away the more likely the detector will not find anymore corners. Actually I found the bug and it never appeared in my datasets :dancers: Just fixed it.

Calibration ran through, however I would only trust the rotation matrix. The translation vector is not correct. This is because a good translation estimate needs enough fast and diverse motion.

[image]

urbste, 2023-03-08:
Another tip: If you run the calibration multiple times, you can set --recompute_corners=0. Otherwise the corner extraction will be done every time and that takes forever ;)

ajsnyde, 2023-03-08:
You freakin' rock man! Any way I can buy you a coffee or something?

urbste, 2023-03-08:
Haha no problem. ;-) Actually, with that rotation estimate you can already run VI-SLAM. In my experience the IMU is very close to the projection center in GoPro cameras. Having the translation vector will increase the accuracy maybe a bit, but will not be a deal breaker.

ajsnyde, 2023-03-08:
Eventually I'm going to move to a sliiiightly better IMU for ~300 bucks.

ajsnyde, 2023-03-08:
Nice to see. I'll give those values a shot on some footage I've got.

ajsnyde, 2023-03-09:
Another rookie question, but how would one turn this:
```Final camera calibration reprojection error: 0.31385 from 66 view.
Focal Length:841.945px Principal Point: 967.348/834.301px.
FISHEYE model: Radial distortion 1: 0.0485829 Radial distortion 2: 0.0209498 Radial distortion 3: 0.0145283 Radial distortion 4: -0.0188735
Finished camera calibration.
```

into an ORBSLAM yaml:
[image]

Would the camera fx and fy be the same focal length? Am I right to assume that the 1-4 radial distortion numbers are k1-k4? Am I using the right Camera.type for a fisheye action cam?

urbste, 2023-03-09:
If you have an aspect ratio calibrated then fx!=fy: `K = [fx 0 cx; 0 a*fy cy;  0 0 1]`. 
Usually the aspect ratio is very small, as pixels should generally be square. Your values look alright though.

If I remember correctly, the fisheye model is the same and radial distortion 1-4 are equal to k1-k4.

closed 2023-03-21 by ajsnyde