urbste/OpenImuCameraCalibrator #39: Problems in the beginning building: cannot find ' opencv2/aruco/dictionary.hpp '

lrz8, 2024-04-08:
thanks for your job
Here is a problem in the beginning when I try to build this project
OpenImuCameraCalibrator/src/core/camera_calibrator.cc:19:10: fatal error: opencv2/aruco/dictionary.hpp: no that direction or index
   19 | #include <opencv2/aruco/dictionary.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

I want to test some data base on the gopro12, so do you have free time to help me handle with it

urbste, 2024-04-08:
Try installing 4.5.x if you can. Unfortunately OpenCV devs decided to do a lot of refactoring with the Aruco module. It was moved from contrib to core.

lrz8, 2024-04-08:
Ok, I will try it letter on 4.5.x version. Hoping it will safely work. Thanks!

发自我的iPhone