urbste/OpenImuCameraCalibrator #43: mono_inertial_gopro_vi tracking lost.

vvcatstar, 2024-05-31:
I use the https://github.com/urbste/ORB_SLAM3/tree/master code to run orbslam3. But the tracking is lost.
The input video is Open ICC calibration [data](https://drive.google.com/file/d/1XjtUX-4ZI0Ydkd2O3BWnaUzfmzm96He4/view), and the slam setting is [gopro9_wide_setting.yaml](https://github.com/urbste/ORB_SLAM3/blob/master/Examples/Monocular-Inertial/gopro9_wide_setting.yaml) or gopro9_linear_setting.yaml.

I can not get the gopro9 example result. The slam3 log is :
```
Examples/Monocular-Inertial/mono_inertial_gopro_vi Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/gopro9_wide_setting.yaml video.mp4 imu_data.json
```
```
RB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Fisheye
- Image scale: 1
- fx: 294.398
- fy: 294.268
- cx: 320.341
- cy: 184.266
- k1: 0.0532949
- k2: 0.0025238
- k3: 0.0478167
- k4: -0.042767
- fps: 50
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1250
- Scale Levels: 6
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[-0.99996901, 0.00167755, -0.0076600099, -0.00083036401;
 0.0076709599, 0.0066179801, -0.99994898, -0.016419901;
 -0.00162677, -0.99997699, -0.00663064, -0.00567071;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 0.0015 rad/s/sqrt(Hz)
IMU gyro walk: 5e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.017 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.0055 m/s^3/sqrt(Hz)
There are 1 cameras in the atlas
Camera 0 is fisheye
Starting the Viewer
First KF:0; Map init KF:0
New Map created with 406 points
Trying to initialize IMU
Empty frame...
Empty frame...
Empty frame...
Empty frame...
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Video FPS: 59.940059940059932
ORB-SLAM 3 running at: 71.061498183881284 FPS
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Init mbg: -0.049462951922201466
-0.048096000704198311
 -0.30794716759215124
Init mba:  -1.048252110912111e-09
-2.4094664177576587e-09
-2.4174839976317348e-08
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 14
mnInitialFrameId = 0
125 Frames set to lost
First KF:12; Map init KF:0
New Map created with 407 points
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Video FPS: 59.940059940059932
ORB-SLAM 3 running at: 70.934532675495944 FPS
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Init mbg:   0.036148811787984243
-0.0027985820598631977
  0.014331851673705209
Init mba: 2.5153335930231403e-08
-1.079735859818857e-08
2.8651734883282952e-08
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 142
mnInitialFrameId = 138
250 Frames set to lost
First KF:23; Map init KF:12
New Map created with 370 points
Trying to initialize IMU
Trying to initialize IMU
Video FPS: 59.940059940059932
ORB-SLAM 3 running at: 69.572878877722673 FPS
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Trying to initialize IMU
Init mbg:  -0.010273129165816786
 0.0047093377012380313
-0.0019212091660355615
Init mba: 3.5785226136105223e-08
2.6513270357983744e-09
1.9580616119964778e-08
Video FPS: 59.940059940059932
ORB-SLAM 3 running at: 87.114744841347928 FPS
Not enough motion for initializing. Reseting...
TRACK: Reset map because local mapper set the bad imu flag 
SYSTEM-> Reseting active map in monocular case
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 274
mnInitialFrameId = 266
380 Frames set to lost
First KF:35; Map init KF:23
New Map created with 333 points
Trying to initialize IMU
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 407
mnInitialFrameId = 404
395 Frames set to lost
First KF:37; Map init KF:35
New Map created with 343 points
Trying to initialize IMU
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
```

imJaewooChoi, 2024-06-04:
where did you get "imu_data.json"?

vvcatstar, 2024-06-07:
use this [code](https://github.com/real-stanford/universal_manipulation_interface/blob/main/scripts_slam_pipeline/01_extract_gopro_imu.py) to get imu_data.json

houhaol, 2025-04-22:
I faced similar situations. May I ask if you have solved somehow?

Thanks a lot