urbste/OpenImuCameraCalibrator #44: Question about orbslam3 setting yaml

imJaewooChoi, 2024-06-03:
I try to use Gopro10 black. 
i run python script file(run_gopro_calibration.py), and i get some result json files.

anyway, i have to setup my setting yaml for orbslam3.
where i can find IMU.NoiseGyro, IMU.NoiseAcc, IMU.GyroWalk, IMU.AccWalk, IMU.Frequency values?

Thanks

urbste, 2024-06-03:
Please try [this](https://github.com/urbste/OpenImuCameraCalibrator/blob/master/docs/imu_noise_parameters.md).
Record a >2h video with your camera. It's best to cover the lens (black screen) so that the video does not get too big.
Also choose the lowest resoultion and fps settings.

imJaewooChoi, 2024-06-03:
Thanks. Do i have to run static_multipose_imu_calibartion too? 
i try to run but it said "FileNotFoundError: [Errno 2] No such file or directory: 'node'"

zhiweicai07, 2025-03-25:
have you solved it

zhiweicai07, 2025-03-25:
> Thanks. Do i have to run static_multipose_imu_calibartion too?谢谢。我是否也必须运行 static_multipose_imu_calibartion？ i try to run but it said "FileNotFoundError: [Errno 2] No such file or directory: 'node'"我尝试运行但它说“FileNotFoundError：[Errno 2] 没有这样的文件或目录：'node'” i think i need javascript files...我想我需要 javascript 文件...

have you solved it ？

zhiweicai07, 2025-03-25:
> Thanks. Do i have to run static_multipose_imu_calibartion too? i try to run but it said "FileNotFoundError: [Errno 2] No such file or directory: 'node'" i think i need javascript files...

have you solved it ？