urbste/OpenImuCameraCalibrator #45: can this tool use to Calibrate insta360 camera?

XUTONG1024, 2024-06-25:


urbste, 2024-07-09:
At the moment, there is no real sperical camera model implemented in pytheiasfm: [camera types](https://github.com/urbste/pyTheiaSfM/tree/master/src/theia/sfm/camera). However, I might actually implement it soon, as I am also interested in calibration 360 cameras. As soon as this is done, we would just have to extend this part here:
https://github.com/urbste/OpenImuCameraCalibrator/blob/a35d40ac3018a2d702badf0b8c87291855a7c14b/include/OpenCameraCalibrator/basalt_spline/ceres_calib_split_residuals.h#L366

XUTONG1024, 2024-07-11:
> At the moment, there is no real sperical camera model implemented in pytheiasfm: [camera types](https://github.com/urbste/pyTheiaSfM/tree/master/src/theia/sfm/camera). However, I might actually [...]

Thank for your reply.I would like to know if Insta360 cameras can utilize their built-in IMU to implement some kind of VIO algorithm, similar to how you implemented ORB-SLAM3 with a GoPro. If you are also interested, I look forward to hearing about your future results.

urbste, 2024-07-11:
I am pretty sure it is possible. You just have to find out how to extract the IMU data from the INSTA360. Is that possible? For GoPro it is fortunately pretty straightforward using the gpmf parser.

XUTONG1024, 2024-07-11:
Yes, I've been able to extract IMU data from insv (Insta360's panoramic video format). I've processed the data, including unit conversions for acceleration, but I've encountered failures when running VINS Mono and ORB-SLAM3. I converted insv files to planar video using Insta Studio, calibrated with Kalibr's pinhole model, and performed joint calibration for IMU noise and camera-IMU alignment, but I'm unsure where the issue lies. If you're interested, I'd be glad to provide you with a package for extracting IMU data in the future.

Richard-coder, 2024-07-23:
@XUTONG1024 , Hello, I am also trying to run VIO using data from a panoramic camera. Would you be willing to share the software package for extracting IMU data from the insv file? I'm curious about how you handle timestamp alignment and unit conversion. I am quite familiar with VIO and am just starting to learn how to process insv data.

XUTONG1024, 2024-07-23:
> @XUTONG1024 , Hello, I am also trying to run VIO using data from a panoramic camera. Would you be willing to share the software package for extracting IMU data from the insv file? I'm curious about [...]

you have to apply for insta360 sdk in insta360 official website, i have not figure out how to align the timestamps between camera and imu, unit conversion is easy just g to 9.81.if you have any advance ,very glad to communicate with me. i am also doing this 360vio.

varunns, 2024-11-27:
@XUTONG1024 Hi Xutong thanks for the previous description. I was wondering if you succeeded in extracting IMU data using insta360 sdk. Please le me know. Thank you!!

XUTONG1024, 2024-11-28:
yes

011218301-徐彤
***@***.***

varunkudan, 2026-01-09:
@Richard-coder @avicomac @urbste I used telemetry-parser to extract IMU data from insv files and used it for other applications. Although I am now trying to do the same for calibration. I was wondering if any of you have made progress towards insta360 calibration for extrinsics between cameras and IMU. Thank you!!