urbste/OpenImuCameraCalibrator #46: Assertion `m_value.object->find(key) != m_value.object->end()' failed

HaomingSong, 2024-06-27:
Hi [urbste](https://github.com/urbste),
Thanks for your awesome work. I'm calibrating a GoPro9 camra with  [OpenImuCameraCalibrator](https://github.com/urbste/OpenImuCameraCalibrator) but failed. I carefully review the related issues but still strugging. I use the provided [borad image](https://github.com/urbste/OpenImuCameraCalibrator/blob/master/resource/board.png) and all the GoPro video can be download from [here](https://drive.google.com/file/d/1H6yJ0xEe8P1NEdfEAfZJvjqRpLEcTr_2/view?usp=drive_link). This is my bash file:
```
DOWNSAMPLE_FAC=4
CAM_MODEL=FISHEYE
CHECKERSIZE=0.022
NUM_SQ_X=10
NUM_SQ_Y=8
CALIB_LINE_DELAY=0
RECOMP_CORNERS=0
GRID_SIZE=0.03

BASEPATH=/OpenImuCameraCalibrator
SETTING=.
DATASET=DATA2

python3 ../python/run_gopro_calibration.py --path_calib_dataset=${BASEPATH}/${SETTING}/${DATASET} \
    --image_downsample_factor=${DOWNSAMPLE_FAC} \
    --camera_model=${CAM_MODEL} \
    --checker_size_m=${CHECKERSIZE} \
    --num_squares_x=${NUM_SQ_X} --num_squares_y=${NUM_SQ_Y} \
    --calib_cam_line_delay=${CALIB_LINE_DELAY} \
    --board_type=apriltag \
    --recompute_corners=${RECOMP_CORNERS} \
    --optimize_board_points=1 \
    --reestimate_bias_spline_opt=1 \
    --voxel_grid_size=${GRID_SIZE}
```
The exception errors:
[image]
 Could you give any advices? Many thanks!

wushandinghua, 2024-08-19:
I had the same problem when calibrated gopro9.
calibrate_camera: /home/***/projects/OpenImuCameraCalibrator/include/OpenCameraCalibrator/utils/json.h:17346: const value_type& nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>::operator[](T*) const [with T = const char; ObjectType = std::map; ArrayType = std::vector; StringType = std::__cxx11::basic_string<char>; BooleanType = bool; NumberIntegerType = long int; NumberUnsignedType = long unsigned int; NumberFloatType = double; AllocatorType = std::allocator; JSONSerializer = nlohmann::adl_serializer; nlohmann::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer>::const_reference = const nlohmann::basic_json<>&]: Assertion `m_value.object->find(key) != m_value.object->end()' failed.
Execute shell:
python python/run_gopro_calibration.py --path_calib_dataset=/home/***/datasets/camera_calibrate/ --checker_size_m=0.015 --path_to_build=/home/***/projects/OpenImuCameraCalibrator/build/applications
@urbste

wushandinghua, 2024-08-19:
@HaomingSong Have you solved yet?