urbste/OpenImuCameraCalibrator #50: Usage with ORBSLAM3

RIPGuy, 2024-08-27:
Hello, I got the results from the calibration thanks to your repo.

I am trying to get the parameters to create a .yaml config for ORBSLAM3.

How do I get the coefficients something like this [https://github.com/urbste/ORB_SLAM3/blob/master/Examples/Monocular-Inertial/gopro9_wide_setting.yaml](url) if I have an output of my results in the camera calibration like this? 

**{
  "final_reproj_error": 0.4680723123482971,
  "fps": 29.97002997002997,
  "image_height": 540,
  "image_width": 960,
  "intrinsic_type": "DIVISION_UNDISTORTION",
  "intrinsics": {
    "aspect_ratio": 1.0151692796284786,
    "div_undist_distortion": 6.477186344000926e-07,
    "focal_length": 648.9719606689928,
    "principal_pt_x": 481.39044108629906,
    "principal_pt_y": 270.0444019781233,
    "skew": 0.0
  },
  "nr_calib_images": 106,
  "stabelized": false
}**

urbste, 2024-08-28:
First you should choose the "FISHEYE" camera model not the DIVISION_UNDISTORTION model for calibration.

Camera.cx and Camera.cy are the principal_pt_x and principal_pt_y.
Camera.fx is the focal_length and Camera.fy is the focal_length*aspect_ratio

For the FISHEYE model you will also get 4 distortion coefficients that you need to set to Camera.k1 to Camera.k4.

RIPGuy, 2024-08-28:
Thank you for responding, I appreciate it. 

I did as you said by using FISHEYE and multiplying focal_length and aspect ratio to get Camera.fy.

Now, I have another problem. I dont exactly know what is the output of the IMU noise and walk calibration. I extracted the telemetry from each video and merged them to a large file as .json and ran fit_allan_variance.

How exaclty am I going to determine Noisegyro, Noiseacc, Walkgyro, Walkacc as something similar in your .yaml config [here](https://github.com/RIPGuy/ORB_SLAM3/blob/master/Examples/Monocular-Inertial/gopro9_wide_setting.yaml).

This is my output:
Loading datastructes
acc_x  num of Cluster 10000
acc_y  num of Cluster 10000
acc_z  num of Cluster 10000
gyr_x  num of Cluster 10000
gyr_y  num of Cluster 10000
gyr_z  num of Cluster 10000
gyr_x  numData 1634680
gyr_x  start_t 0.10694
gyr_x  end_t 8145.25
gyr_x dt 
-------------8145.14 s
-------------135.752 min
-------------2.26254 h
gyr_x  freq 200.694
gyr_x  period 0.00498272
gyr_y  numData 1634680
gyr_y  start_t 0.10694
gyr_y  end_t 8145.25
gyr_y dt 
-------------8145.14 s
-------------135.752 min
-------------2.26254 h
gyr_y  freq 200.694
gyr_y  period 0.00498272
gyr_z  numData 1634680
gyr_z  start_t 0.10694
gyr_z  end_t 8145.25
gyr_z dt 
-------------8145.14 s
-------------135.752 min
-------------2.26254 h
gyr_z  freq 200.694
gyr_z  period 0.00498272
Gyro X 
C -104.081  1177.53  -28.969 -4.71871 0.143624
 Bias Instability 4.10462e-05 rad/s
 Bias Instability 0.000140574 rad/s, at 679.891 s
 White Noise 147.062 rad/s
 White Noise 0.0408523 rad/s
  bias 0.00967336 degree/s
-------------------
Gyro y 
C  -34.1982   438.615  -109.522    8.4892 -0.132931
 Bias Instability 0.000121343 rad/s
 Bias Instability 0.000101923 rad/s, at 247.247 s
 White Noise 35.3665 rad/s
 White Noise 0.00995467 rad/s
  bias 0.058988 degree/s
-------------------
Gyro z 
C   -54.3667    685.758   -74.3565    6.14682 -0.0952854
 Bias Instability 0.000214828 rad/s
 Bias Instability 0.000159052 rad/s, at 1999.48 s
 White Noise 113.224 rad/s
 White Noise 0.0315557 rad/s
  bias 0.103114 degree/s
-------------------
==============================================
==============================================
acc_x  numData 1634680
acc_x  start_t 0.10694
acc_x  end_t 8145.25
acc_x dt 
-------------8145.14 s
-------------135.752 min
-------------2.26254 h
acc_x  freq 200.694
acc_x  period 0.00498272
acc_y  numData 1634680
acc_y  start_t 0.10694
acc_y  end_t 8145.25
acc_y dt 
-------------8145.14 s
-------------135.752 min
-------------2.26254 h
acc_y  freq 200.694
acc_y  period 0.00498272
acc_z  numData 1634680
acc_z  start_t 0.10694
acc_z  end_t 8145.25
acc_z dt 
-------------8145.14 s
-------------135.752 min
-------------2.26254 h
acc_z  freq 200.694
acc_z  period 0.00498272
acc X 
C -0.000312564   0.00433467  -0.00424375   0.00631176 -0.000131427
 Bias Instability 0.00624745 m/s^2
 White Noise 0.0896063 m/s^2
-------------------
acc y 
C  -0.00546312    0.0440721  -0.00664157   0.00870627 -0.000180689
 Bias Instability 0.0254901 m/s^2
 White Noise 0.710141 m/s^2
-------------------
acc z 
C  -0.00053097   0.00492891     0.001008    0.0005775 -1.23887e-05
 Bias Instability 0.00309473 m/s^2
 White Noise 0.0947435 m/s^2
-------------------

Could you shed light on how I should proceed with this?

frank-cole, 2024-09-02:
@RIPGuy Did you know how obtain Noisegyro, Noiseacc, Walkgyro, Walkacc?

RIPGuy, 2024-09-02:
Hello, I'm not entirely sure if I got it right but the Random Noise should
correspond to the White Noise and Random Walk should refer to the Bias
Instability of each gyro and acc.

The Acc's result is already there but with the Gyr's result, I think you
should select the instability or noise with the lowest value since there
are two values. I based this idea by looking on some of the .yaml config
files they almost usually select the one with the lowest value.

I averaged all x-y-z and got a single value for all noisegyro, noiseacc,
accwalk, and accgyro.

Maybe, if by any chance, you found another way this can be understood
please inform me. Thank you, hoped I helped.

frank-cole, 2024-09-03:
> Hello, I'm not entirely sure if I got it right but the Random Noise should correspond to the White Noise and Random Walk should refer to the Bias Instability of each gyro and acc. The Acc's result [...]

Thinks, if i figure out, i will share with you.

frank-cole, 2024-09-03:
> Hello, I'm not entirely sure if I got it right but the Random Noise should correspond to the White Noise and Random Walk should refer to the Bias Instability of each gyro and acc. The Acc's result [...]

Thanks, if i figure it out, i will share it with you.

RIPGuy, 2024-10-14:
Sorry for reviving this post, I just have one concern. I read something about turning off the hypersmooth setting as it crops the view which makes it difficult to run the SLAM in visual-inertial setting. I just wondered if had you turned it off when conducting your calibration dataset?