urbste/OpenImuCameraCalibrator #57: static_imu_calib dataset link in imu_intrinsics.md is invalid

JafarAbdi, 2025-06-22:
I'm trying to run the example in https://github.com/urbste/OpenImuCameraCalibrator/blob/master/docs/imu_intrinsics.md, but looks like the link for the static_imu_calib dataset is no longer valid.

JafarAbdi, 2025-06-26:
> Hey, you are right I removed it by accident. I would upload a new one but I currently do not have access to my camera.

Thanks for your response!

> Do you need guidance how to record one on your own?

That would be amazing, thanks!

Clemadril, 2025-07-10:
@urbste [static_multipose_imu_calibration.py](https://github.com/urbste/OpenImuCameraCalibrator/blob/master/python/static_multipose_imu_calibration.py) is still referencing javascript implementation for telemetry extraction in master branch where it has been replaced by your py_gpmf python package.